Doodle3D/print3d #15: Monitor SD Card prints

peteruithoven, 2015-03-09:
Currently we won't get information on prints that are started using the SD card.
This might be a useful addition. 
Marlin does support M-codes to request status from or control SD card prints:
https://github.com/MarlinFirmware/Marlin/blob/Development/Documentation/GCodes.md#m-codes

oliv3r, 2015-03-10:
Mind you, that we won't be using sd, display or user-input via marlin in the 2.1 printer. All of this will be directly driven via the olimex pins.

peteruithoven, 2015-03-10:
So the Olimex will become responsible for reading SD cards?

oliv3r, 2015-03-10:
yep

olijf, 2016-03-11:
Seems not all 3D printer firmware support pushing to SD card from the serial line. My prusa i3 sunhokey reprap supports listing/pushing to sd card...

peteruithoven, 2016-03-11:
To clarify, this is about monitoring SD card prints, not about putting G-Code on a SD-card through usb. 
This has low priority.