Doodle3D/print3d #16: OpenWRT compile error

peteruithoven, 2015-05-15:
When compiling the develop branch version in openwrt the following line triggers an error: 
https://github.com/Doodle3D/print3d/blob/develop/src/settings.h#L13
It was added in the following commit:
https://github.com/Doodle3D/print3d/commit/d3c7d3c3853eaf47a5f748aff13c1793491013cb

arneboon, 2015-05-26:
https://github.com/Doodle3D/print3d/commit/987254860785ec83ee1904c63f05703df9bd1066

disabled error

arneboon, 2015-05-26:
compiling develop branch...

arneboon, 2015-05-26:
```
make[2]: Leaving directory '/home/openwrt/trunk'
make[1]: Leaving directory '/home/openwrt/trunk'
[ok] Build successful
> copy bin folder to shared folder
> path: /home/openwrt/shared/bin
> complete
[ok] new built files are copied succesfully
[ok] done.
```

success!

arneboon, 2015-05-26:
flashed firmware to device

can not connect to lulzbot taz 4 printer

```
root@wifibox:/# tail -f tmp/print3d-ttyACM0.log
26-05 15:33:24  [SER] setSpeed(): 115200
26-05 15:33:24  [ABD] setState(): 2:connecting > 2:connecting
26-05 15:33:25  [CDH] received get state command
26-05 15:33:29  [CDH] received get state command
26-05 15:33:30  [SER] setSpeed(): 250000
26-05 15:33:30  [ABD] setState(): 2:connecting > 2:connecting
26-05 15:33:33  [CDH] received get state command
26-05 15:33:36  [SER] setSpeed(): 115200
26-05 15:33:36  [ABD] setState(): 2:connecting > 2:connecting
26-05 15:33:37  [CDH] received get state command
26-05 15:33:41  [CDH] received get state command
26-05 15:33:42  [SER] setSpeed(): 250000
26-05 15:33:42  [ABD] setState(): 2:connecting > 2:connecting
26-05 15:33:45  [CDH] received get state command
26-05 15:33:48  [SER] setSpeed(): 115200
26-05 15:33:48  [ABD] setState(): 2:connecting > 2:connecting
26-05 15:33:49  [CDH] received get state command
26-05 15:33:53  [CDH] received get state command
```

keeps looping connection cycle

peteruithoven, 2015-05-26:
Please enabling more verbose logging. This is done by specifying `-V` instead of `-v` when print3d is started. This is usually set in [print3d-runner](https://github.com/Doodle3D/print3d/blob/master/src/script/print3d-runner.sh). (Also mentioned in http://www.doodle3d.com/help/debugging )

And, someone put that error there for a reason, let's figure that out as well.

woutgg, 2015-05-28:
Hmm this had to do with compilation targets. The commit in which it was added seems to be about support for non-wrt linux. Two possibilities as I see it:
1. it was a miner's canary to detect an (invalid) linux+uci combination (but it is not conditional on presence of openwrt so that does not make sense), or
2. it was left there by accident...but then how come the issue never showed before?

In any way, afaik this can be safely removed. In worst case it will trigger a compilation error on the uci.h include.

arneboon, 2015-05-31:
hi wouter thanks for your response

works great now, lets close this issue

closed 2015-05-31 by arneboon