Doodle3D/print3d #19: p3d gives wrong error when print3d is not started

casperlamboo, 2015-06-01:
when communicating with p3d and print3d is not started p3d gives the following error

```
01-06 16:53:14  could not connect domain socket with path '/tmp/print3d-tty.usbmodem1421' (Connection refused)
error: could not open socket (run in verbose mode for details)
```

Expected error "Print3D is not started"

olijf, 2016-03-11:
Detecting whether or not Print3D is running might not be that easy as "error: could not open socket" is somewhat vague.
p3d has to be run as sudo on some operating systems if print3d has been started as sudo (if no rules are configured in udev)  so running p3d as non sudo gives the same error.

peteruithoven, 2016-03-11:
Valid point, that error can't be mapped to one issue. Let's close this.

closed 2016-03-11 by peteruithoven