Doodle3D/print3d #29: print3d does not start when image is build using Image Generator

companje, 2015-11-20:
I found this comment in the print3d Makefile on [line 18](https://github.com/Doodle3D/print3d/blob/master/Makefile#L18).
`#Note: as mentioned in http://wiki.openwrt.org/doc/devel/dependencies, the inotifyd dep will not be checked on installation through opkg`

I'm trying to install the Doodle3D packages using OpenWrt's Image Generator and I think the problems I face (print3d does not get started) are related to this.

Any clues?

companje, 2015-11-20:
btw this is my Image Generator build command:
```make image PROFILE=TLMR3020 PACKAGES="wifibox -ppp -kmod-ipv6 -6relayd -kmod-ip6tables -libip6tc -odhcp6c"```

peteruithoven, 2015-11-23:
I would start by checking which of the startup scripts get executed, see: https://github.com/Doodle3D/print3d#startup-scripts