Doodle3D/print3d #30: M115 for printer detection

companje, 2015-12-08:
Currently M105 (get temperature) is used to detect successful communication with a printer.
Would it be possible and more useful to use M115 for this. This gives a lot more info like printer name/type etc.

For example:
```
SENDING:M115
ok PROTOCOL_VERSION:0.1 FIRMWARE_VERSION:1.1.7477 FIRMWARE_NAME:pr3Dator MACHINE_TYPE:CraftBot EXTRUDER_COUNT:1
```

olijf, 2016-03-11:
Seems my prusa i3 sends back 
```
https://www.sunhokey.com Machine:Prusa I3
ok
```
So not much info is given. (Also the ok comes after the response)

peteruithoven, 2016-03-11:
This would be useful to automatically adjust machine settings like heated bed support, bed size etc right? But seeing that, for example the Ultimaker 2 Go doesn't mention it being the 2 go version of a Ultimaker 2 I'm unsure this will be useful.

closed 2016-03-11 by peteruithoven