Doodle3D/print3d #36: Craftbot: print3d hangs after M104 because the response is interpreted as a M105 response

companje, 2016-02-15:
The Craftbot firmware has some small differences causing print3d to hang. For example "M104 S200" returns ok wíth temperature which is different from Marlin. Here's (http://reprap.org/wiki/G-code#M104:_Set_Extruder_Temperature) a description of M104 on the RepRap site.

olijf, 2016-02-15:
After sending `M104 S180` at the end of the printing process it seems to stop te printer somehow. This only happens at the end of a print. 
Disabling the line in you printer config (in the advanced part of the printer config webpage) seems to be an easy fix and workaround.

Relevant part of the log
`15-02 14:50:18  [ABD] printNextLine(): 2903/2906
15-02 14:50:18  [MLD] sendCode(): M104 S180
15-02 14:50:18  [MLD] extractGCodeInfo()
15-02 14:50:18  [MLD]   (extractGCodeInfo) targetTemperature_: 180
15-02 14:50:18  [MLD] readCode(): 'ok T:230'
15-02 14:50:18  [MLD] parseTemperatures(): 'ok T:230'
15-02 14:50:18  [MLD]   checkTemperatureInterval_: '5000'
15-02 14:50:20  [MLD] update temp()
15-02 14:50:20  [MLD]   check temp
15-02 14:50:20  [MLD] sendCode(): M105
15-02 14:50:20  [MLD] extractGCodeInfo()
15-02 14:50:20  [MLD] readCode(): 'ok T:230 B:70'` 

Print3D seems to think the response `ok T:230` is a reaction to a `M105` command resulting in not returning the control to the users, forcing him to restart his doodl3d box.

Please advice users about this current workaround.

peteruithoven, 2016-02-15:
But why doesn't this happen with the start gcode? This should also contain a temperature change command.

companje, 2016-02-15:
My theory is that when it just misses 1 normal 'ok' (because of the
temperature behind it) it has still enough gcode commands in the queue.
Only at the end of a print it gets stuck because it keeps waiting because
it is expecting one more 'ok' but it never arrives.

peteruithoven, 2016-02-15:
But, there is no such queue? It only advances to the next line when it receives an ok. If we miss one ok it should just stop sending lines. I don't understand what triggers print3d to send the next line.

companje, 2016-02-15:
I think you're right Peter. It was the prototype where send 5 lines at the
start without waiting for ok to and then each next line after receiving an
ok. But in the MarlinDriver (
https://github.com/Doodle3D/print3d/blob/master/src/drivers/MarlinDriver.cpp) I
don't see this.

olijf, 2016-02-15:
In https://github.com/Doodle3D/doodle3d-firmware/blob/master/src/conf_defaults.lua#L225 there does not seem to be a `M104` in the start code.

olijf, 2016-02-16:
Attached is the log of sevaral sucessful prints. @peteruithoven wanted to checkout the `Q128` code as response to `M109`
[craftbot.txt](https://github.com/Doodle3D/print3d/files/132285/craftbot.txt)

olijf, 2016-02-16:
As requested the relevant part of the log.
```
15-02 16:01:01  [MLD] readCode(): 'ok Q:128'
15-02 16:01:01  [ABD] printNextLine(): 2/331
15-02 16:01:01  [MLD] sendCode(): G21 
15-02 16:01:01  [MLD] extractGCodeInfo()
15-02 16:01:01  [MLD] readCode(): 'ok Q:128'
15-02 16:01:01  [ABD] printNextLine(): 3/331
15-02 16:01:01  [MLD] sendCode(): G91 
15-02 16:01:01  [MLD] extractGCodeInfo()
15-02 16:01:01  [MLD] readCode(): 'ok Q:128'
15-02 16:01:01  [ABD] printNextLine(): 4/331
15-02 16:01:01  [MLD] sendCode(): M107 
15-02 16:01:01  [MLD] extractGCodeInfo()
15-02 16:01:01  [MLD] readCode(): 'ok Q:128'
15-02 16:01:01  [ABD] printNextLine(): 5/331
15-02 16:01:01  [MLD] sendCode(): G28 X0 Y0 
15-02 16:01:01  [MLD] extractGCodeInfo()
15-02 16:01:02  [MLD] readCode(): 'ok Q:128'
15-02 16:01:02  [ABD] printNextLine(): 6/331
15-02 16:01:02  [MLD] sendCode(): G28 Z0 
15-02 16:01:02  [MLD] extractGCodeInfo()
15-02 16:01:03  [SRV] read 0 bytes from client with fd 5
15-02 16:01:03  [SRV] read 4 bytes from client with fd 5
15-02 16:01:03  [IPC] command: [>>getState]
15-02 16:01:03  [CDH] received get state command
15-02 16:01:03  [IPC] command: [<<ok]
15-02 16:01:03  [SRV] read 0 bytes from client with fd 5
15-02 16:01:03  [SRV] read 10 bytes from client with fd 5
15-02 16:01:03  [IPC] command: [>>getTemperature:s1]
15-02 16:01:03  [CDH] received get temperature command with arg 1
15-02 16:01:03  [IPC] command: [<<ok]
15-02 16:01:03  [SRV] read 10 bytes from client with fd 5
15-02 16:01:03  [IPC] command: [>>getTemperature:s2]
15-02 16:01:03  [CDH] received get temperature command with arg 2
15-02 16:01:03  [IPC] command: [<<ok]
15-02 16:01:03  [SRV] read 10 bytes from client with fd 5
15-02 16:01:03  [IPC] command: [>>getTemperature:s3]
15-02 16:01:03  [CDH] received get temperature command with arg 3
15-02 16:01:03  [IPC] command: [<<ok]
15-02 16:01:03  [SRV] read 10 bytes from client with fd 5
15-02 16:01:03  [IPC] command: [>>getTemperature:s4]
15-02 16:01:03  [CDH] received get temperature command with arg 4
15-02 16:01:03  [IPC] command: [<<ok]
15-02 16:01:03  [SRV] read 10 bytes from client with fd 5
15-02 16:01:03  [IPC] command: [>>getTemperature:s5]
15-02 16:01:03  [CDH] received get temperature command with arg 5
15-02 16:01:03  [IPC] command: [<<ok]
15-02 16:01:03  [SRV] read 0 bytes from client with fd 5
15-02 16:01:03  [SRV] read 4 bytes from client with fd 5
15-02 16:01:03  [IPC] command: [>>getProgress]
15-02 16:01:03  [CDH] received get progress command
15-02 16:01:03  [IPC] command: [<<ok]
15-02 16:01:03  [MLD] update temp()
15-02 16:01:03  [MLD]   check temp
15-02 16:01:03  [MLD] sendCode(): M105
15-02 16:01:03  [MLD] extractGCodeInfo()
15-02 16:01:03  [MLD] readCode(): 'ok T:228 B:63'
15-02 16:01:03  [MLD] parseTemperatures(): 'ok T:228 B:63'
15-02 16:01:03  [MLD]   checkTemperatureInterval_: '5000'
15-02 16:01:04  [MLD] readCode(): 'ok Q:128'
15-02 16:01:04  [ABD] printNextLine(): 7/331
15-02 16:01:04  [MLD] sendCode(): G1 Z15 F9000 
15-02 16:01:04  [MLD] extractGCodeInfo()
15-02 16:01:06  [MLD] readCode(): 'ok Q:128'
15-02 16:01:06  [ABD] printNextLine(): 8/331
15-02 16:01:06  [MLD] sendCode(): G92 E0 
15-02 16:01:06  [MLD] extractGCodeInfo()
15-02 16:01:06  [MLD] readCode(): 'ok Q:128'
15-02 16:01:06  [ABD] printNextLine(): 9/331
[... 214 lines cut ...]
15-02 15:53:05  [IPC] command: [>>getState]
15-02 15:53:05  [CDH] received get state command
15-02 15:53:05  [IPC] command: [<<ok]
15-02 15:53:06  [MLD] readCode(): 'ok Q:128'
15-02 15:53:06  [ABD] printNextLine(): 1999/2003
15-02 15:53:06  [MLD] sendCode(): M84 
15-02 15:53:06  [MLD] extractGCodeInfo()
15-02 15:53:06  [MLD] readCode(): 'ok Q:128'
15-02 15:53:06  [ABD] printNextLine(): 2000/2003
15-02 15:53:06  [MLD] sendCode(): G90 
15-02 15:53:06  [MLD] extractGCodeInfo()
15-02 15:53:06  [MLD] readCode(): 'ok Q:128'
15-02 15:53:06  [ABD] printNextLine(): 2001/2003
15-02 15:53:06  [MLD] sendCode(): M140 S70
15-02 15:53:06  [MLD] extractGCodeInfo()
15-02 15:53:06  [MLD]   (extractGCodeInfo) targetBedTemperature_: 70
15-02 15:53:06  [MLD] readCode(): 'ok B:70'
15-02 15:53:06  [ABD] printNextLine(): 2002/2003
15-02 15:53:06  [MLD] sendCode(): M117 Done                 
15-02 15:53:06  [MLD] extractGCodeInfo()
15-02 15:53:06  [MLD] readCode(): 'ok Q:128'
15-02 15:53:06  [ABD] printNextLine(): 2003/2003
15-02 15:53:06  [ABD] resetPrint()
15-02 15:53:06  [ABD] setState(): 5:printing > 3:idle
```

peteruithoven, 2016-02-17:
## Craftbot
(Using Pronterface)
### Marlin default start gcode
**Set nozzle target temperature**
`M109 S{printingTemp}`
Response: `ok Q:128`
**Set target bed temperature**
`M190 S{printingBedTemp}`
Response: `ok Q:128`

### marlin default end gcode
**Set nozzle target temperature**
`M104 S{preheatTemp}`
No response...
**Set target bed temperature**
`M140 S60`
Response: `ok B:50`
(Looks like it responds with current bed temperature)

### Marlin preheat / heatup command: 
Same g-code as start gcode.

Regulier commando, zoals `G21`: 
Response: `ok Q:128`

## Ultimaker2
(Using p3d on WiFi-Box because of baudrate)
`G21`
Response: `ok`
`M109 S180`
Response: `ok`
M104 S180
Response: `ok`

So sending `M104 S...` (Set extruder temperature (not waiting)) doesn't give a response, even though sending a `M140 S...` does. We need to figure out how to cooldown this printer.

peteruithoven, 2016-02-17:
Quick options: 
- Putting that  M104 command at the end. But then it sticks in printing state which means you can only press stop. After pressing stop it sticks in stopping mode. So not really an option. 
- Using waiting command (M109) instead, users will have to wait for it to cooldown to preheat temp before it can start heating up for a next print. 

I'll check if the Craftbot software uses a special cooldown gcode command.

peteruithoven, 2016-02-17:
The g-code from CraftWare ends with:
```
G21 ; mm
G90 ; abs
G92 E0 ; reset extr
M106 S255
G0 F4800 X38.071 Y106.733

G0X0Y0

M104 S0 ;turn off head heater
M140 S0 ;turn off bed heater
M106 S0
```
So this uses a regular `M104 S...`(The M106 is the fabspeed)
It's also followed by `;CraftWare_Settings=...` but that isn't relevant.

peteruithoven, 2016-02-17:
Sending `M104 S0` also gives no response.

companje, 2016-02-17:
A workaround for now would be to use an M109 S180 in Doodle3D's end-code for the CraftBot.
I will contact CraftUnique so they can fix the issue of the lacking OK response after M104.

companje, 2016-02-19:
I got a response from CraftUnique. There experiment shows that M104 S180 returns 'ok T:21' which is not according to the RepRap specs but still it does return something. I confirmed this by connecting the CraftBot to my Mac and opening the Arduino IDE's Serial Monitor. It gives this response indeed when using the correct line-endings (LF).

companje, 2016-02-19:
Interesting: Sending an `M104 S180` from `Printrun` does not give a response.
Sending the same from Arduino (or from Processing) returns `ok T:21`. The response I receive in Processing is correctly followed by a line-feed.

peteruithoven, 2016-02-19:
Can you see in Processing whether the response line-feed is the same after a M104 and a M115 for example?

companje, 2016-02-20:
The response ends in both cases with a line-feed (ASCII character 0x0A)

olijf, 2016-02-23:
As the response to an `M104` (`ok T:230`) is different than that of an `M105` (`ok T:69 B:23`) it might be possible to create a fix for the craftbot.
```
15-02 14:50:18  [MLD] sendCode(): M104 S180
15-02 14:50:18  [MLD] readCode(): 'ok T:230'
[...]
23-02 15:56:14  [MLD] sendCode(): M105
23-02 15:56:14  [MLD] readCode(): 'ok T:69 B:23'
```

peteruithoven, 2016-02-23:
We've also confirmed that we do receive a response from the Craftbot when sending a `M104`.

peteruithoven, 2016-02-23:
Currently, in the marlin driver, we have 2 communication "lines" to a printer. One in which we occasionally check the temperature and one in which we send the buffered gcode commands. See our [drivers docs](https://github.com/Doodle3D/print3d/blob/master/docs/drivers.md) for more info. 
To support responses from commands other than temperature checks that look like temperature checks, wouldn't it be possible to use one communication line? Currently we send temperature check gcode directly, we'd have to add those to the beginning of the gcode buffer. 
Currently we only send the next line when it's not a temperature response, and the response begins with `ok`, we'd have to change that into sending the next line when it begins with `ok` (which also includes the temperature responses) and parse the temperature when that's also included in that response. 

Where we parse responses: 
https://github.com/Doodle3D/print3d/blob/master/src/drivers/MarlinDriver.cpp#L128-L175
Sending temperature requests:
https://github.com/Doodle3D/print3d/blob/master/src/drivers/MarlinDriver.cpp#L228-L230
Appending gcode (to the back of the buffer):
https://github.com/Doodle3D/print3d/blob/master/src/drivers/MarlinDriver.cpp#L123-L126
https://github.com/Doodle3D/print3d/blob/master/src/drivers/AbstractDriver.cpp#L85-L89

It's probably useful to build on the upcoming gcode buffer improvements, since includes a lot of gcode buffer changes and improvements. 
There are happening in the following branch: https://github.com/Doodle3D/print3d/tree/feature/logging%2Bchecking-renewed

peteruithoven, 2017-01-13:
We did do a commit where we added custom endcode for the craftbot, where we replaced the M104 with a M109. But as mentioned above this is a waiting / blocking command, so the user can only do something else after is has reached the preheat temperature. 
https://github.com/Doodle3D/doodle3d-firmware/commit/1cb130965977ecd2de5beb1de226a40b287769bd