Doodle3D/print3d #39: Crash on start print when logging verbose

peteruithoven, 2016-02-23:
When I put print3d in verbose logging connect the printer and start a new print it crashes. So up until starting a print it goes okay. This doesn't happen when loglevel is `info`. 
It doesn't show a reason in the print3d logs, logread doesn't show an error. 
We're investigating the core dump.

peteruithoven, 2016-02-23:
Print 3D logs:
```
23-02 11:36:34 [MAIN] (info)   : starting Print3D server
23-02 11:36:34 [MAIN] (info)   : using printer device: 'ttyACM0'
23-02 11:36:34 [MAIN] (info)   : using printer type from UCI configuration
23-02 11:36:34 [MAIN] (info)   : using log level: verbose
23-02 11:36:34 [SRV ] (info)   : printer type from settings: 'ultimaker2go'
23-02 11:36:34 [DFCT] (verbose):   num drivers: 2
23-02 11:36:34 [DFCT] (info)   : Created firmware: ultimaker2go
23-02 11:36:34 [GCB ] (verbose): init - max size: 3072.0KiB, bucket size: 50.0KiB, split size: 8.0KiB
23-02 11:36:34 [SRV ] (info)   : starting printserver with socket path '/tmp/print3d-ttyACM0'
23-02 11:36:34 [ABSD] (info)   : openConnection()
23-02 11:36:34 [ABSD] (info)   :   serial port path: '/dev/ttyACM0', baudrate: 115200
23-02 11:36:34 [SER ] (info)   : open(): /dev/ttyACM0
23-02 11:36:34 [SER ] (info)   : setSpeed(): 115200
23-02 11:36:34 [ABSD] (info)   : setState(): 1:disconnected > 2:connecting
23-02 11:36:34 [MLND] (verbose): (checking connection) check temperature 0/2
23-02 11:36:36 [MLND] (verbose): (checking connection) check temperature 1/2
23-02 11:36:38 [SER ] (info)   : setSpeed(): 250000
23-02 11:36:38 [ABSD] (info)   : setState(): 2:connecting > 2:connecting
23-02 11:36:40 [MLND] (verbose): (checking connection) check temperature 0/2
23-02 11:36:41 [ABSD] (info)   : setState(): 2:connecting > 3:idle
23-02 11:36:45 [SRV ] (verbose): cancelling all transactions except for fd 7 (#clients: 1)
23-02 11:36:45 [GCB ] (verbose): clear
23-02 11:36:45 [GCB ] (verbose): append() - len: 21291, excerpt: ';Generated', seq_num: 0, seq_ttl: 6, src: 10.0.0.132; ttl size (lines): 0 (0); totalLines arg: 2883
23-02 11:36:45 [GCB ] (verbose): append() - added 21291 bytes of gcode (3 chunks) in 48 ms
23-02 11:36:45 [ABSD] (verbose):   targetTemperature_: 230
23-02 11:36:45 [ABSD] (verbose):   targetBedTemperature_: 70
23-02 11:36:45 [ABSD] (info)   : setState(): 3:idle > 4:buffering
23-02 11:36:45 [ABSD] (verbose): startPrint
23-02 11:36:45 [ABSD] (verbose): resetPrint()
23-02 11:36:45 [ABSD] (info)   : setState(): 4:buffering > 3:idle
23-02 11:36:45 [ABSD] (info)   : setState(): 3:idle > 5:printing
```

woutgg, 2016-02-23:
Information on inspecting an openwrt core dump [can be found here](http://www.doodle3d.com/help/wiki/analyzing-uhttpd-crashes).

peteruithoven, 2016-02-23:
Was fixed with: https://github.com/Doodle3D/print3d/commit/3b94280d3e63b5c144dfdd88ab01f55ca1c5e08f

closed 2016-02-23 by peteruithoven