Doodle3D/print3d #40: Stopping makerbot print stays in stopping state

woutgg, 2016-02-23:
Makerbot printers (including Wanhao duplicator 4) have an internal 512 byte buffer which (almost) never empties when the stop button is clicked, causing the print server to stay in stopping state indefinitely. This problem (usually) does not occur when a print finishes.

To investigate this, we could add a mechanism to the driver to remember the data that has been sent but is still in the printer's buffer. We then migth be able to use the s3g decompiler to find out what the printer is doing or waiting for.

peteruithoven, 2016-04-04:
Could we also try removing parts of the end gcode, seeing if removing a certain line makes a difference?

woutgg, 2016-04-04:
Yes, that is definitely also an option and might be easier too, at least to start with.

peteruithoven, 2016-04-18:
I see we did try removing some stop gcode:
https://github.com/Doodle3D/doodle3d-client/issues/226#issuecomment-46839796

peteruithoven, 2016-04-19:
Jeroen found that removing the `M73`'s from the end gcode seems to solve the issue:
https://github.com/Doodle3D/WiFi-Box/issues/5#issuecomment-211799970

peteruithoven, 2016-05-11:
Fixed with: https://github.com/Doodle3D/doodle3d-firmware/commit/65dd10b43cebc5e45b2de8796fc485d6415724ee

closed 2016-05-11 by peteruithoven